peplin/pygatt #263: Supported scenarios for reconnect?

rogerdahl, 2019-12-12:
First, thank you for this library!

I've found that I'm unable to reconnect when the device (a smart watch) disconnects. The disconnect event is:

```
DEBUG    {'after': b'\n\x1b[0;94m[fd:d1:c7:18:70:2d]\x1b[0m[LE]> \r\n(gatttool:10674):'
DEBUG              b' GLib-\x1b[1;33mWARNING\x1b[0m **: \x1b[34m00:18:22.898\x1b[0m: '
DEBUG              b'Invalid file descriptor',
DEBUG     'before': b'',
DEBUG     'callback': [<bound method GATTToolBackend._disconnect of <pygatt.backends.gatttool.gatttool.GATTToolBackend object at 0x7ffad2d98a90>>,
DEBUG                  functools.partial(<function disconnect_callback at 0x7ffad2d8d400>, <__main__.Uwatch2 object at 0x7ffacf27a6d8>)],
DEBUG     'event': <threading.Event object at 0x7ffad2dbcd68>,
DEBUG     'match': <_sre.SRE_Match object; span=(0, 131), match=b'\n\x1b[0;94m[fd:d1:c7:18:70:2d]\x1b[0m[LE]> \r\n>,
DEBUG     'patterns': ['.*Disconnected', '.*Invalid file descriptor']}
```

Should I be able to use `GATTToolBackend.reconnect()` in this case?

Passing `auto_reconnect=True` to `GATTToolBackend.connect()` works, but keeps the connection constantly open. I think I'd prefer to reconnect only when and if the user issues a new command to be sent to the watch.

peplin, 2021-02-21:
Hey there! Better late than never.

`GATTToolBackend.reconnect()` is an internal function for the backend and isn't useful for this purpose. It's a helper method to handle auto reconnect, when that's enabled. 

When a device disconnect is detected, the knowledge of the device is removed from the backend. You should be able to reconnect by calling `connect(...)` again with the device MAC.

closed 2021-02-21 by peplin

rogerdahl, 2021-02-22:
Haha, yes, it's a while since I worked on that. I was never able to get things working reliably. The basic issue seemed to be that after some random failed connection, the system ended up in a bad state that I couldn't recover from. I ended up trying stuff like automatically resetting and re-adding BT adapters, even automatically killing and restarting various processes. I blamed it all on Bluez.