peplin/pygatt #269: How to know what charasteristic are a handle?

lrmoreno007, 2020-01-29:
Hello! I'm connecting to a BLE Smart Remote https://www.amazon.es/CCsky-Bluetooth-Wireless-SmartRemote-Compatible/dp/B07BTSHKCG and I discover this charasteristics with the example in this page https://pypi.org/project/pygatt/ :
```
Read UUID 00002a4d-0000-1000-8000-00805f9b34fb: b'0000000000000000'
Read UUID 00002a4c-0000-1000-8000-00805f9b34fb: b'00'
Read UUID 00002a19-0000-1000-8000-00805f9b34fb: b'64'
Read UUID 00002a01-0000-1000-8000-00805f9b34fb: b'c403'
Read UUID 00002a4a-0000-1000-8000-00805f9b34fb: b'11010001'
Read UUID 00002a4b-0000-1000-8000-00805f9b34fb: b'05010906a1018501050719e029e71500250175'
Read UUID 00002a00-0000-1000-8000-00805f9b34fb: b'536d61727452656d6f7465'
Read UUID 00002a04-0000-1000-8000-00805f9b34fb: b'1e003c000400e803'
Read UUID 00002a50-0000-1000-8000-00805f9b34fb: b'12324456781001'
Read UUID 00002a4e-0000-1000-8000-00805f9b34fb: b'01'
Read UUID 00002a05-0000-1000-8000-00805f9b34fb: b'00000000'
```

When i start my code (example in the same page) and subscribe to `00002a4d-0000-1000-8000-00805f9b34fb` ( I thing it's correct, but not sure) and activate debug, I receive this:
```
DEBUG:pygatt.backends.gatttool.gatttool:gatttool_cmd=gatttool -i hci0 -I
INFO:pygatt.backends.gatttool.gatttool:Running...
INFO:pygatt.backends.gatttool.gatttool:Connecting to FF:FF:80:01:46:A8 with timeout=5.0
DEBUG:pygatt.device:Looking up handle for characteristic 00002a4d-0000-1000-8000-00805f9b34fb
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a05-0000-1000-8000-00805f9b34fb, value handle: 0x3
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a00-0000-1000-8000-00805f9b34fb, value handle: 0x7
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a01-0000-1000-8000-00805f9b34fb, value handle: 0x9
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a04-0000-1000-8000-00805f9b34fb, value handle: 0xb
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a50-0000-1000-8000-00805f9b34fb, value handle: 0xe
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4e-0000-1000-8000-00805f9b34fb, value handle: 0x11
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4d-0000-1000-8000-00805f9b34fb, value handle: 0x13
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4d-0000-1000-8000-00805f9b34fb, value handle: 0x17
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4d-0000-1000-8000-00805f9b34fb, value handle: 0x1b
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4d-0000-1000-8000-00805f9b34fb, value handle: 0x1f
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4d-0000-1000-8000-00805f9b34fb, value handle: 0x23
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4b-0000-1000-8000-00805f9b34fb, value handle: 0x26
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4a-0000-1000-8000-00805f9b34fb, value handle: 0x28
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a4c-0000-1000-8000-00805f9b34fb, value handle: 0x2a
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a19-0000-1000-8000-00805f9b34fb, value handle: 0x2d
DEBUG:pygatt.device:Found <Characteristic uuid=00002a4d-0000-1000-8000-00805f9b34fb handle=35>
DEBUG:pygatt.backends.gatttool.gatttool:Sending cmd=char-write-req 0x24 0100

********AT THIS TIME THE DEVICE DISCONNECT*************

ERROR:pygatt.backends.gatttool.gatttool:No response received
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 574, in char_write_handle
    self.sendline(cmd)
  File "/usr/lib/python3.5/contextlib.py", line 66, in __exit__
    next(self.gen)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 191, in event
    self.wait(event, timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 157, in wait
    raise NotificationTimeout()
pygatt.exceptions.NotificationTimeout: None
Traceback (most recent call last):
  File "Subscribe.py", line 30, in <module>
    device.subscribe(uuid4, callback=handle_data)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/device.py", line 222, in subscribe
    wait_for_response=wait_for_response
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/device.py", line 17, in wrapper
    return func(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/device.py", line 48, in char_write_handle
    self._backend.char_write_handle(self, handle, *args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 50, in wrapper
    return func(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 574, in char_write_handle
    self.sendline(cmd)
  File "/usr/lib/python3.5/contextlib.py", line 66, in __exit__
    next(self.gen)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 191, in event
    self.wait(event, timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 157, in wait
    raise NotificationTimeout()
pygatt.exceptions.NotificationTimeout: None
```
I thing that the timeout trigger because the device disconnect in the point `DEBUG:pygatt.backends.gatttool.gatttool:Sending cmd=char-write-req 0x24 0100`

If in this timeout I push a button, I receive:
```
INFO:pygatt.device:Received notification on handle=0x17, value=0xb'ea00'
INFO:pygatt.device:Received notification on handle=0x17, value=0xb'0000'
```
But Callback never trigger.

When I subscribe to other charasteristic like 00002a19-0000-1000-8000-00805f9b34fb, there are no problem whit timeout and see the INFO of button pressed, but callback don't occurs too as expected by not being subscribed to the correct UUID.

Looking to this, I see that the charasteristic are repeated with many handles and the handle 0x17 never are subscribed/notificated, and there are some problem writing to the charasteristic that make my device disconnect.

Help please, thanks in advance.

Makalov, 2020-02-13:
I also had a problem with notification callback never being called. I think this is a problem with gatttool. The callback function only triggers when I perform another pygatt read.  

https://stackoverflow.com/questions/58420137/gatttool-stop-receiving-notification-after-seconds-not-because-of-disconnection

Jabberwacky, 2022-09-16:
As I heard, pygatt is unable to process several instances of one characteristic marked with the different handles. Probable, this is the reason.