peplin/pygatt #274: bgapi scan not working under Linux

CLDiego, 2020-03-05:
I recently acquired a BLED112 to see if it was more reliable than my normal Bluetooth dongle. I've tried starting the BLED112 with pygatt and connecting and seems to work fine (I still need to test subscribing to any characteristics).

However, I'm getting the following error when trying to scan for devices:

pygatt.backends.bgapi.exceptions.ExpectedResponseTimeout: Timed out after 1.000000s waiting for [<ResponsePacketType.gap_set_scan_parameters: 64>].

This under Ubuntu, I get the same error using python2 and 3. I tried using the logger but it says the same thing. 

I tried the dongle on windows 10 and it works fine.

akshmakov, 2020-03-27:
I am experiencing this as well. 

We are trying to work on cross-platform tidy up of https://github.com/neurotechx/eeg-notebooks which depends on pygatt through muse-lsl, when testing with BGAPI Backend this issue crops up on Ubuntu 18.04 systems

Some progress that I have done, It does appear to be system specific. Changing the default timeout in `BGAPIBackend.expect_any` (https://github.com/peplin/pygatt/blob/master/pygatt/backends/bgapi/bgapi.py#L599)  to 10 seconds allowed succesful scan, and itnermitent connection, failing on a similar timeout in a different location.

In any case while I try to dig further wanted to give a heads up, this feels to me as platform serial port issue and perhaps a change in BLED1112 firmware.

Perhaps the read loop in bgapi (https://github.com/peplin/pygatt/blob/master/pygatt/backends/bgapi/bgapi.py#L640-L659) can be improved to buffer the raw ser.read ... perhaps messages are being missed ?

peplin, 2021-02-21:
@akshmakov your theory about a system-specific serial port issue, or a firmware change, sounds reasonable. I tried but unfortunately I'm not able to trigger this bug with my BLED112. I would feel uncomfortable messing with the read loop, but if you ended up getting this working I'd be happy to merge a PR.

closed 2021-02-21 by peplin