peplin/pygatt #280: No module named 'pygatt'

meluvalli, 2020-05-01:
All steps done under Windows 10 as Administrator Command Prompt.

Steps:
Installed python 3.8
Installed pygatt using command "pip install pygatt"
tried test.py and get error "No module named 'pygatt'"

<pre>C:\>pip install pygatt
Requirement already satisfied: pygatt in c:\program files (x86)\python38-32\lib\site-packages (4.0.5)
Requirement already satisfied: pyserial in c:\program files (x86)\python38-32\lib\site-packages (from pygatt) (3.4)
Requirement already satisfied: enum-compat in c:\program files (x86)\python38-32\lib\site-packages (from pygatt) (0.0.3)</pre>

<pre>C:\>test.py
Traceback (most recent call last):
  File "C:\Users\Aaron\test.py", line 1, in <module>
    import pygatt
ModuleNotFoundError: No module named 'pygatt'
</pre>

Test.py contents:
<pre>
import pygatt

adapter = pygatt.BGAPIBackend()
adapter.start() #start your dongle
adapter.scan()
</pre>

meluvalli, 2020-05-01:
Update: 
I downloaded the GIT repository and and manually installed it using setup.py install.

This fixed the issue.  So, it appears just can't use pip install command.  Had to manually download and install it.

peplin, 2021-02-20:
I'm glad you got it working. I'm afraid I don't have a Windows development environment to test with anymore.

closed 2021-02-20 by peplin