peplin/pygatt #281: Use with built in bluetooth on macOS?

niemyjski, 2020-05-01:
I have a bluetooth LE device that is paired with my MacBook Pro via browser/os. But I cannot get this library to run and detect the same device no matter what I try. To me it seems as if I am required to use a bluetooth dongle from reading all the threads? What do I need to do to use the built in bluetooth? I've tried a bunch of stuff. 

```
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/cu.Bluetooth-Incoming-Port - n/a
Traceback (most recent call last):
```

m9h, 2020-12-06:
Unfortunately you will need the BLED112 adaptor on Mac OS X. If you have found an alternative solution I would be very interested to know more about it.

peplin, 2021-02-20:
Sorry, I don't have a Mac and don't know much about its Bluetooth subsystem.

closed 2021-02-20 by peplin