peplin/pygatt #282: pygatt.scan request sudo

RogersNelson, 2020-05-02:
Hi,

I would like to make a LE scan without any user action.
So I've done :
```
sudo   setcap 'cap_net_raw,cap_net_admin+eip' `which hcitool\``
```

And this is my python code: 

```
import pygatt
adapter = pygatt.GATTToolBackend()`
adapter.start()
print("Scan BLE devices...")
ble_nearby_devices = adapter.scan()
print(ble_nearby_devices)
```

It still request sudo password:
```[sudo] Mot de passe de Nelson: 
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    adapter.start()
  File "/usr/local/lib/python3.8/dist-packages/pygatt/backends/gatttool/gatttool.py", line 264, in start
    self.reset()
  File "/usr/local/lib/python3.8/dist-packages/pygatt/backends/gatttool/gatttool.py", line 631, in reset
    subprocess.Popen(["sudo", "systemctl", "restart", "bluetooth"]).wait()
  File "/usr/lib/python3.8/subprocess.py", line 1079, in wait
    return self._wait(timeout=timeout)
  File "/usr/lib/python3.8/subprocess.py", line 1804, in _wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.8/subprocess.py", line 1762, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt
```

Despite this:
```
$ sudo getcap /usr/bin/hcitool 
/usr/bin/hcitool = cap_net_admin,cap_net_raw+eip
$ sudo getcap `which hcitool`
/usr/bin/hcitool = cap_net_admin,cap_net_raw+eip
```

Python 3.8.2
Debian bullseye/sid
gattlib==0.20200122
pygatt==4.0.5

I would appreciate any help.
Cheers,
Nelson

RogersNelson, 2020-05-02:
Now, i have to un

peplin, 2021-02-21:
I think you found the right solution - disabling the rest on start. You are correct to point out that the docs are incomplete here, as the gatttool backend does use `sudo` for this restart in addition to the hcitool commmand.

Since gatttool is deprecated by BlueZ, I'm inclined to leave this as-is and add a note to the comment. Thanks for the report.

PR #308 (2021-02-21): Note a required workaround for gatttool with sudoless operation
Fixes #282
commit 538c24f3 (2021-02-21): Note a required workaround for gatttool with sudoless operation

Fixes #282
closed 2021-02-21 by peplin