peplin/pygatt #285: char_write_long issue

iscapla, 2020-05-08:
Hello,
I'm not really sure that I've been using the char_write_long function properly. In my code, a single char_write works fine but the data transmitted is concatenated to MTU length (for some reason, my device only accepts 23 bytes at the write operation, notifications received are bigger than that).

Could you please provide a simple example about how to write long buffers into one characteristic?
I though that the solution was to use char_write_long function but the library throw a NotImplementedError exception.

Am I doing something wrong?

A possible example (from the API examples) could be?

```
import pygatt
from binascii import hexlify

adapter = pygatt.GATTToolBackend()

def handle_data(handle, value):
    """
    handle -- integer, characteristic read handle the data was received on
    value -- bytearray, the data returned in the notification
    """
    print("Received data: %s" % hexlify(value))

try:
    adapter.start()
    device = adapter.connect('01:23:45:67:89:ab')

    device.subscribe("uuid1",
                     callback=handle_data)

    device.char_write_long("uuid2", data_buffer_with_240_bytes, wait_for_response=True)

finally:
    adapter.stop()
```

In other issue I read that this function was introduced in v4.0 and my library version is v4.0.5; running the code in a Raspberry Pi 3 Model B Plus Rev 1.3.

Thanks so much.

lunakoa, 2020-06-28:
Wouldnt the UUID be long, something like 00002a06-0000-1000-8000-00805f9b34fb?

iscapla, 2020-09-21:
Hello @lunakoa,
Yes, I didn't write a correct UUID for the example.

Btw, any news about this issue?

peplin, 2021-02-21:
The GATTTool backend unfortunately does not support long writes. That's only supported by the BGAPI backend.

closed 2021-02-21 by peplin