peplin/pygatt #289: Connect failing using gatttool backend

victorromeo, 2020-06-25:
```
Ubuntu 20.04 x86_64
gattlib==0.20200122
pygatt==4.0.5
pexpect==4.6.0
```

Hi,

I'm having trouble connecting, regardless of the connection mode (random or public), to a nRF52840 Arduino Nano 33 BLE Sense, using pygatt.  It appears to be related to the pexpect 'connect' regular expression pattern in my opinion, as I am able to connect immediately via the terminal.  I'm not a regex specialist sorry.  May I get your opinion?

```
ian@henry:~/Documents/PlatformIO/Projects/SoundDetect/pygatt$ sudo python3 ./transfer.py DEBUG:pygatt.backends.gatttool.gatttool:gatttool_cmd=gatttool -i hci0 -I
INFO:pygatt.backends.gatttool.gatttool:Running...
...truncated...
INFO:pygatt.backends.gatttool.gatttool:Discovered DE:D9:D3:AC:75:B9 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered name of DE:D9:D3:AC:75:B9 as AudioTransfer
...truncated...
INFO:pygatt.backends.gatttool.gatttool:Found 13 BLE devices
.
Device UUID: DE:D9:D3:AC:75:B9
INFO:pygatt.backends.gatttool.gatttool:Connecting to DE:D9:D3:AC:75:B9 with timeout=10
DEBUG:pygatt.backends.gatttool.gatttool:sendline: sec-level low
DEBUG:pygatt.backends.gatttool.gatttool:sendline: connect DE:D9:D3:AC:75:B9 random
ERROR:pygatt.backends.gatttool.gatttool:Timed out connecting to DE:D9:D3:AC:75:B9 after 10 seconds.
Disconnected from BLE DE:D9:D3:AC:75:B9
DEBUG:pygatt.backends.gatttool.gatttool:sendline: disconnect
INFO:pygatt.backends.gatttool.gatttool:Stopping
DEBUG:pygatt.backends.gatttool.gatttool:sendline: exit
```
When using gatttool manually via the terminal, the connection is established immediately.
```
ian@henry:~/Documents/PlatformIO/Projects/SoundDetect/pygatt$ gatttool -i hci0 -I
[                 ][LE]> connect DE:D9:D3:AC:75:B9 10
Attempting to connect to DE:D9:D3:AC:75:B9
Connection successful
[DE:D9:D3:AC:75:B9][LE]> 
```

Connect as random does not issue Connection successful
```
ian@henry:~/Documents/PlatformIO/Projects/SoundDetect/pygatt$ gatttool -i hci0 -I
[                 ][LE]> connect DE:D9:D3:AC:75:B9 random
Attempting to connect to DE:D9:D3:AC:75:B9
[DE:D9:D3:AC:75:B9][LE]> 
```
Connection as public does issue Connection successful
```
[                 ][LE]> connect DE:D9:D3:AC:75:B9 public
Attempting to connect to DE:D9:D3:AC:75:B9
Connection successful
[DE:D9:D3:AC:75:B9][LE]>
```

heykube, 2020-10-18:
I'm seeing the same behavior on the Raspberry Pi 4 (BlueZ 5.50), and pulling pygatt from Master. 

I can easily connect using gatttool on the command line, but pygatt gives a timeout error.  I'm writing the firmware for the BTLE device so I actually know that it's connecting immediately, but pygatt comes back with a timeout error.  If I retry the connect 4 times in a row, eventually it recognizes the connection and then works well.

heykube, 2020-10-18:
```
    def connect(self, address, timeout=DEFAULT_CONNECT_TIMEOUT_S,
                address_type=BLEAddressType.public, auto_reconnect=False):
        log.info('Connecting to %s with timeout=%s', address, timeout)
        self.sendline('sec-level low')
        **time.sleep(0.1)**
        self._address = address
        self._auto_reconnect = auto_reconnect
```
I downloaded the library, and I think it has to do with sending the 'sec-level low' command, and not waiting for the response before sending the "connect" command. If I delay just a bit, I can get the connection to work each time. 

I guess the proper way to build the Pexect with the sec-level command first.

mitea1, 2020-11-05:
@heykube thanks a lot. can confirm this fixes the issue.

however there's a typo in your post: its `echo 1000 > /sys/kernel/debug/bluetooth/hci0/supervision_timeout`

commit 0a63c2db (2021-02-21): Mention increasing kernel timeout for gatttool

Fixes #289
PR #313 (2021-02-21): Mention increasing kernel timeout for gatttool
Fixes #289
commit c5de2f87 (2021-02-21): Mention increasing kernel timeout for gatttool

Fixes #289
closed 2021-02-21 by peplin