peplin/pygatt #300: BGAPIBackend window size to search for more UUIDs

fcondo, 2020-10-31:
Hello, I have an Arduino Nano 33 BLE with 3 characteristics to which I want to subscribe, on Windows. But when I run `device.discover_characteristics()` I only found the default characteristics:
`INFO:pygatt.backends.bgapi.bgapi:Characteristic 0x00002a00-0000-1000-8000-00805f9b34fb is handle 0x3
INFO:pygatt.backends.bgapi.bgapi:Characteristic 0x00002a01-0000-1000-8000-00805f9b34fb is handle 0x5`

According to [BLE specs](https://specificationrefs.bluetooth.com/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf) they are related to device name and appearance. 

With my phone I am able to see the characteristics with the correspondent UUIDs, but the discovering on Windows does not, seems like there is a limit on the search windows size. Any suggestion?

CLDiego, 2020-11-01:
I'm not familiar with the Nano 33 BLE, but it sounds like it could be: 

 - The service might be private or non-default. You could see on the datasheet if this is the case and if so it should be made public just by changing a flag on your Nano.
 - your central BLE (your PC) might have trouble with the advertisements. I'm not sure if you are connecting to the device or just scanning. If you are just scanning to discover characteristics then increasing the scan time might help, or configuring your Nano to increase the advertisement interval.
 - If you are connecting to it, then rather than discovering the characteristics you could try using the UUID directly.

Hope it helps!

fcondo, 2020-11-01:
Hello Diego, thanks for your support. This is what I tried:

- I am able to connect to the device. Then I tried to connect both using subscribe and char_read directly to the UUIDs. The problem occurs when trying to subscribe to the characteristics with Notify enabled, no one is found. 

- I set random UUIDs on 128 bit by using an online generator for both BLE Service and for BLECharacteristic. May this affect the result?

- I am using nRF Connect on my Android phone and I am able to see the characteristics (see [here](https://ibb.co/v3dbXDp)). The flags are [these](https://ibb.co/JKgKXdb) one, I have not found clear documentation on the meaning of the flags. But the phone does not keep alive the connection. It drops after some seconds and I have to reconnect. 

Any other suggestions?

peplin, 2021-02-21:
If you are still having issues, I would recommend running a test without the random UUIDs to try and narrow down the cause. This may have to do with some limitation of the BLED112 itself, but I don't have any evidence. Sorry I can't be of help, I've not seen this particular issue on my test devices.

closed 2021-02-21 by peplin