peplin/pygatt #304: Consider asyncio bleak backend

xloem, 2021-01-06:
https://github.com/hbldh/bleak is a cross-platform bluetooth interface using python's asyncio infrastructure.  The use of asyncio provides for more flexibility for the user.  bleak has native python implementations for linux, windows, and osx, requiring no calling out to other processes or asking the user for root access.

xloem, 2021-01-11:
I basically implemented this here: https://github.com/alexandrebarachant/muse-lsl/pull/148/files

Here it is with the pump loop automatically run in a separate thread: https://github.com/alexandrebarachant/muse-lsl/pull/148/commits/6336053d644c0149e641c82ec13751e46398faf3

peplin, 2021-02-20:
Bleak looks great! @xloem I'm curious, what value do you see in adding PyGATT as a wrapper? I'm not actively maintaining this project anymore and haven't found anyone interested in picking it up, so I'm wondering if the best course of action is to consider it deprecated and redirect people to bleak instead.

xloem, 2021-02-21:
I just did that to reduce the problems introduced by changing everything.  I don't know if one of the pygatt backend is really important for another operating system, or if people have work or plans based around them.  I tried to make the bleak addition small and optional in case there was any issue with it.

Yuribtr, 2021-07-17:
Well, I tried Bleak for some months in production (connection to GoPro cameras) and can say that I saw strange behavior of BleakClient. It's method write_gatt_char doesn't guarantee that command was successfully send to device. BleakClient needs about 10-30 retries to send command. Sometimes it is doesn't work at all, but native gattool console utility works like a charm. So I decided to try pygatt.  
At present I have scanning, pairing and trust functions made with bluetoothctl, the last part will be sending commands function via pygatt, and Bleak will be dropped out from my project.

xloem, 2021-07-17:
Hey that's a great report but why are you mentioning it here instead of the bleak project?  Did you open an issue so they can learn of it and fix it?

Yuribtr, 2021-07-17:
> Hey that's a great report but why are you mentioning it here instead of the bleak project? Did you open an issue so they can learn of it and fix it?

Yep, this is good idea, maybe I will have time to describe this bug with enough details.

> EDIT: I see you are saying pygatt still has value. 

Correct! This was my main intention to say that pygatt might be useful in some scenarios.