peplin/pygatt #327: Trying to subscribe...

aldocarrolo, 2021-06-02:
Hi, i'm trying to use pygatt to connect my BLE device to my raspberry pi but i'm having an issue. The problem is that when i subscribe the raspberry to a service of my BLE device this one connects and after one second it disconnects (it doesn't mantain connection with subscription's function). 

This is the code i use

```
import time

import pygatt
from binascii import hexlify

adapter = pygatt.GATTToolBackend()

def handle_data(handle, value):
    """
    handle -- integer, characteristic read handle the data was received on
    value -- bytearray, the data returned in the notification
    """
    print("Received data: %s" % hexlify(value))

adapter.start()
device = adapter.connect('D8:9E:61:F1:9B:74')

device.subscribe("00002a19-0000-1000-8000-00805f9b34fb",
                     callback=handle_data)

adapter.stop()
```

Before using this module, I used gatttool to verifying the correct functionality of BLE device (and it worked).

P.S: I'm emulating the BLE device with smartphone

traversjames, 2021-06-03:
Duplicate of #326

peplin, 2021-06-10:
I added a basic method of blocking the main thread to the docs: #330

closed 2021-06-10 by peplin